davidkingsman/split-flap #2: Unit Frame Assembly Missing Holes for M3 nuts

ThomDyson, 2021-07-14:
First, thank you so much for this project. I've had to learn Arduino and how to order custom PCBs, so this great.

In the instructions, section 5.2.2 says "Push two M3 nuts diagonally in the back of the frame. Use a screw to pull the nut if necessary."

I think this is to mount the stepper motor, but there are no holes in the unit frame for the M3 nuts under the stepper motor.

Dave19171, 2021-07-14:
This is not to mount the stepper motor, you need M4 screws for that. These are for attaching the frame to the HousingFrame
[image]

ThomDyson, 2021-07-14:
Got it. Thanks.

closed 2021-07-14 by ThomDyson