davidkingsman/split-flap #3: CAP-SMD_L3.2-W1.6 What part?

TheBekker, 2021-07-16:
Can you help specifying the "CAP-SMD_L3.2-W1.6" part?

[image]

Dave19171, 2021-07-16:
Are you aware, that you need only one PCB with all the parts? Read the instructions on page 7 for details. There is also a table with each part on page 5.

I used this capacitor originally: https://lcsc.com/product-detail/Tantalum-Capacitors_TORCH-CA45-A-10V-22%CE%BCF-K_C503893.html
It does not seem to be in stock anymore.
The data sheet for the AMS1117-3.3 says: "The circuit design used in the AMS1117 series requires the use of
an output capacitor as part of the device frequency compensation. The addition of 22µF solid tantalum on the output will ensure
stability for all operating conditions. "

Since you only need one, I suggest you get one elsewhere and solder it yourself.  You need a 
22uF >5V tantalum capacitor in 0805 size.

TheBekker, 2021-07-16:
Ahh i just saw that the instructions are only on PrusaPrinters :) 
Bit confusing that half is here and other ther :)

I'll check the instructions, thanks!

Dave19171, 2021-07-16:
Just the instructions for the code on here. Everything electromechanical is on prusaprinters.

closed 2021-07-16 by TheBekker