davidkingsman/split-flap #4: ESP01 communication with unit not working / how to debug

rkuerbitz, 2021-08-07:
Dear David, first of all thanks for sharing your great design! I am in progress of making a version with laser-cut flaps out of 0.8 mm black acrylic with cut-plot letters - I will share those files when I've got them working.

However, I can't get the ESP01 communicate with the unit. I debugged the unit and put in some additional output via serial console and have the unit working properly and go to all letters fine. I put in some more serial message into unit to see if unit gets any messages from the ESP - receiveLetter in the unit never fires.

So perhaps you do have a hint on how to debug the ESP01; I am not sure how to see the output of #serial on the IDE (while it is plugged into the unit board). 

Thanks in advance for any hints.

Dave19171, 2021-08-07:
Hey, 
the debugging of the ESP01 is indeed not easy when it is plugged in on the board. My solution was to use a NodeMCU with a USB port and just plug in the SDA, SCL and GND pins with jumper wires. 

But regarding your problem, if the slaves do not react and the ESP sends a message, it is most likely an I2C issue.
Try to confirm that the hardware is correctly assembled:

- You soldered the two BSS138 (Q1, Q2) correctly to your master board
- The four 10K pull-ups (R1-R4) are in place  
- These components are just on the master board with the ESP01

If your hardware is assembled correctly, you need a logic analyzer to troubleshoot the I2C bus. There are cheap units under $10 available.

rkuerbitz, 2021-08-10:
Dear David, thanks for the hints. I was able to debug the Master Code using a NodeMCU board as suggested and I found the error: I hadn't adapted the UNITSAMOUNT when testing with my first 2 assembled units and thus got stuck in the error loop. Everything is working perfectly now.

I am currently working on a 3 rows x 16 columns display using 3 masters. I'll try to adapt the Master code a little using D12 and D13 on the units as Row ID.

Dave19171, 2021-08-10:
I would recommend adapting the code and saving the unit's information (Address and in your case row) in the EEPROM. That way you could have as many units as you like (or as there are I2C addresses, which is ~128).

rkuerbitz, 2021-08-10:
Good idea using the EEPROM. I will publish my code when I'm done (will take some weeks...). Thanks again.

closed 2021-08-15 by Dave19171