davidkingsman/split-flap #5: unit.ino not compiling when #define test uncommented

sergios100, 2021-08-11:
Hi Dave,

First let me congratulate you for that exellent project! 
I am still building the clock, as parts arrive step by step from China.
I tried to upload the unit.ino file to my chinese Arduino Nanos, which worked after I switched to "ATmega328P (Old Bootloader) in "Tools" -> "Processor".
Upload went fine. But as soon as I uncomment Line 6: "#define test" it will not compile anymore and is throwing the following error:

```
Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano, ATmega328P (Old Bootloader)"
...\split-flap-master\unit\unit.ino: In function 'void loop()':

unit:134:36: error: cannot convert 'String' to 'int' for argument '1' to 'void rotateToLetter(int)'

     rotateToLetter(currentCalLetter);

                                ^

exit status 1

cannot convert 'String' to 'int' for argument '1' to 'void rotateToLetter(int)'
```
Seems to be some sort of type conversion error.
Could you have a look at it? I don't get it at the moment. It seems that you want to be able to show these sequence: " ", "Z", "A", "U", "N", "?", "0", "1", "2", "9" but on the other hand the function rotateToLetter() seems to only take in integers (probably the numer of the tile which contains the desired letter? 

Best Regards and many thanks!

Sergio

Dave19171, 2021-08-11:
You are totally correct, that test routine is from before I switched to integers for I2C communication. Characters were too unreliable. I will change the code accordingly.

ThomDyson, 2021-08-11:
Sergio,

Try changing line 131 from this 

  String calLetters[10] = {" ", "Z", "A", "U", "N", "?", "0", "1", "2", "9"};

to  something like
 int calLetters[10[ = { 1, 26, 2, 20, 31, 32, 33, 34, 35, 39};

I don't know what letters that will produce, but it should solve the type problem.

Dave19171, 2021-08-11:
Fixed the issue by updating the test routine

closed 2021-08-11 by Dave19171