davidkingsman/split-flap #8: no ip-adress

anno006, 2021-10-16:
Can't find the esp01 within my network. The code uploads without an error. How can i 1. give it a static ip address or 2. what is the problem? also the serial monitor works not.

anno006, 2021-10-16:
> > Did you uncomment line 23 in the ESPMaster.ino? `//#define serial //uncomment for serial debug messages, no serial messages if this whole line is a comment!`
> 
> Yes i did already.

But after rewriting, and rewriting, and etc. i do have now a error state from the serial monitor:

Connecting to WiFi .....scandone
state: 0 -> 2 (b0)
.state: 2 -> 0 (2)
reconnect
...scandone
state: 0 -> 2 (b0)
.state: 2 -> 0 (2)
reconnect
...scandone

anno006, 2021-10-16:
I try, but i get the hole day the error connection refused. That's why i try to find what is wrong, 2 weeks ago it did work.

Dave19171, 2021-10-16:
I can not really help you with that because your network is probably the issue. Here is a tutorial to configure a static IP-address for the library I am using: [https://randomnerdtutorials.com/esp8266-nodemcu-static-fixed-ip-address-arduino/](url)