davidkingsman/split-flap #9: return false missing in FlapFunctions

Klimbim1, 2021-12-26:
Hi Dave,
my compiler from the Arduino IDE moans if the return false; is missing in the translateLetterToInt function is missing.
Mine looks now like this:

```
//translates char to letter position
int translateLettertoInt(char letterchar) {
  for (int i = 0; i < FLAPAMOUNT; i++) {
    if (letterchar == letters[i]) {
      return i;
    }
  }
  return false; //ADDED
}
```

Dave19171, 2022-01-02:
What version of the Arduino IDE are you using? Are you compiling for the board "Generic ESP8266 Module"?

I am trying to replicate the compiler error, but it compiles fine.
Also, this function's return type is `int`, I am confused why there should be a `return false` there.

Klimbim1, 2022-01-03:
I am using 1.8.19 and I did compile for Generic ESP8266 Module. Return type yes you're right, my fault should be return 0 then I guess?

The error message I get is:
`exit status 1
control reaches end of non-void function [-Werror=return-type]`

Klimbim1, 2022-01-10:
Any more ideas, @Dave19171?

closed 2022-06-26 by Dave19171