davidkingsman/split-flap #10: shortage of C5

lewpaw, 2022-02-15:
Hey Dave, thanks for this cool project.
I'm not an electronics expert, especially when it comes to soldering SMT elements, that's why I wanted to order these assembled from JLCPCB. However, they are in shortage of "CA45-A-10V-22μF-K" (C5). Can this be replaced with something similar? Can you recommend?
Thanks!

Dave19171, 2022-02-15:
You can think of SMD capacitors as normal ones. For our use case you only need to match the size (0805), capacity (22μF) and voltage (>=10V). C5 is a tiny bit special because it is a tantalum capacitor.

This seems to be the only equivalent currently in stock on LCSC: [https://lcsc.com/product-detail/Tantalum-Capacitors_ROHM-Semicon-TCTP1A226M8R_C2859526.html](https://lcsc.com/product-detail/Tantalum-Capacitors_ROHM-Semicon-TCTP1A226M8R_C2859526.html)
 
But you only need one for the power supply of the ESP01. 
Maybe try searching on eBay and solder it yourself. If I require just a few of something, I usually find it there.

Hopsen, 2022-04-29:
Hi Dave, i also wanted to use jclpcb and their soldering service. 
At the moment they are short of C2 and U4. Am I right I can replace C2 with "50V 100nF X7R ±10% 0805 Multilayer Ceramic Capacitors MLCC"??

Can U4 be replaced with "Texas Instruments ULN2003ADR"?

Dave19171, 2022-04-29:
> Hi Dave, i also wanted to use jclpcb and their soldering service. At the moment they are short of C2 and U4. Am I right I can replace C2 with "50V 100nF X7R ±10% 0805 Multilayer Ceramic Capacitors MLCC"??

Yes, a 50V cap will work as long as it is also 100nf and a 0805 package.
 
> Can U4 be replaced with "Texas Instruments ULN2003ADR"?
That one should work, it is still a ULN2003A. If I understand the datasheet correctly, it is just packaged differently.

closed 2023-06-13 by Dave19171