davidkingsman/split-flap #11: Connecting and sending first text but not more

diablex2, 2022-06-24:
Hi,
Finished the project. Thank you very much for your effort to share it. Great documentation for a beginner like me.

I can connect to 10 digit unit with ip address of ESP01. I can send the first text message. But when I send a second message nothing happens. When I check the packets sent and received to esp01 from the router I see the packages sent. When I cut the power and power on again all digits turn to show blank flap than I see the last message (the message which lost) I sent before cutting the power.
So, for a new message on split flap I need to cut power and re connect it. 
The clock shows up but it does not changing the minute digit. I think this is also related to my problem.

Could the community please share their opinions about the my problem?

Thank you in advance.

Dave19171, 2022-06-26:
Are you able to uncomment line 5 in one of the units and look at the serial monitor? I suspect it will show that the units are not correctly waking up from sleep.

You can test this by making line 97 to 120 in the unit code a comment. If this fixes it, your units are not correctly waking up.

diablex2, 2022-07-04:
Thank you very much for your attention Dave19171.
I am trying the solution you proposed. As I am working on it in the evenings it will take some time but I will share the result here.

diablex2, 2022-11-24:
> Thank you very much for your attention Dave19171. I am trying the solution you proposed. As I am working on it in the evenings it will take some time but I will share the result here.

Sorry for my late reply.  I tried the solution but not worked. I changed the first unit with a new soldered one.  And everything was working fine.  Thats my case.

closed 2022-11-24 by Dave19171