davidkingsman/split-flap #15: Last 3 digits won´t spin correctly

NoI3ody, 2022-12-04:
Hello,

i have build your amazing splitflap... but i have a problem with the 3 last digits of my splitflap.
As you can see in the video, all flaps are calibrating, after that the 5 middle ones start rotating till they display the correct time. But while the 5 middle ones are rotating, the last 3 start rotating until they get a timeout.
https://youtu.be/CSe2SAsITus 

I have no idea what´s going on here. Tried to switch the nanos, the units, the cables but i couldn´t get it right?
Anyone have a idea what i can test here?

Dave19171, 2022-12-05:
Have you tried enabling the serial output of the nanos and look at that? Maybe there is something going on with the power and not the code.

NoI3ody, 2022-12-09:
I enabled the serial and testet. The last 3 Units gets this code in serial monitor.
Could it be the power? I have a 3A powersource from a LED-strip.

I can´t post the serial monitor. I added a Screenshot of the beginning and the end of the rotation as the content between looks very similar to me.
Beginn of serial monitor:
[image]
End of serial monitor:
[image]

NoI3ody, 2022-12-25:
Any ideas, what the problem could be?

Dave19171, 2022-12-25:
I have no idea. Your serial output looks very weird. It starts formatted correctly, but then everything is in one line. I am still suspecting something with your power because of the gibberish characters in the serial output and the restarting in the middle of a line.

NoI3ody, 2023-01-07:
I found the problem. It was not the powersupply.
It was the ESP. The ESP sends for the last digits -1 when it has no data for the left over units. However my arduinos are converting -1 to 255 as you can see in the first letter received and after receiving 255 the code crashes. So i changed the errorhandling in FlapFunctions.ino Line 52 from -1 to 0. Now when i have an error or blank data it will display the blank tile.
With this change everything is working as it should.

closed 2023-01-07 by NoI3ody